robodhruv/visualnav-transformer #7: where deploy the model?

ForestWang, 2023-10-25:
great work! 
do you deploy the model on embeded board such as Xavier NX or 3090 like GPUs?
Thanks

dimikout3, 2023-11-21:
Hi all! I have also built a robot around a Jetson Xavier NX Developement Kit and I was wondering if anyone has experience in deploying ViNT in such a companion computer.

robodhruv, 2024-01-26:
Hi! Thanks for your interest in our works.
We deploy all models on Jetson computers onboard the robot.

We have tested the following boards and can confirm that all models are able to run at 5Hz or larger on these: Jetson
TX2, Xavier NX, Xavier AGX, Orin NX, Orin AGX.

The code provided in the repo should work out of the box for these devices. Please let me know if you run into any issues.

Dhruv

closed 2024-01-26 by robodhruv