robodhruv/visualnav-transformer #11: What do the four values of choosen_waypoint represent?

Hwh0865, 2024-01-03:
Very nice work! 
When deploying, if choosen_waypoint=[ 0.15498, -0.00129, 0.9395, -0.3425 ], the robot needs linear and angular velocities to be driven, how can I calculate the linear and angular velocities from the four data, and what do these four data mean? 
Thank you!

closed 2024-01-05 by Hwh0865

wmh02240, 2024-01-09:
Hi, have you deployed this model on your robot? How is the effect? I deployed the NoMad model, but the effect is not good, collisions often occur。
Looking forward to your reply。

Hwh0865, 2024-01-09:
Good afternoon! Glad to hear of your successful deployment. I've just successfully deployed ViNT to my robot and am having the same collision problem. The current speculation is that the problem is two-fold, on one hand it may be the number of pictures in the topology map (only nine), and on the second hand the vint model has not been exposed to this unknown environment and is not able to realize the navigation function, it may need to re-collect the data from the current environment and then fine-tune it. Looking forward to your reply, thanks!

wmh02240, 2024-01-12:
Thanks for your reply, I have collected the date from my office and fine-tune the NoMad model , the same problem of collosion also occur. Have you resolve this problem? 
Looking forward to your reply, Thanks.

Hwh0865, 2024-01-18:
哥你好，你是怎么微调的，是采完数据然后训练得到的新模型，就算微调结束了吗，需要有什么注意的么
@wmh02240

robodhruv, 2024-01-27:
https://github.com/robodhruv/visualnav-transformer/issues/13