robodhruv/visualnav-transformer #15: long-distance navigation

Follu-co, 2024-01-26:
Is it possible for the published code to achieve long-distance navigation? Is this the final version of the code? Are you considering releasing the relevant code of the heuristic or diffusion model in the future?

robodhruv, 2024-01-27:
Yes, you should be able to deploy the [NoMaD model](train/vint_train/models/nomad) for exploration and long-distance navigation by sampling from the mode. NoMaD includes a diffusion head over actions and goal masking so it does not need a separate image goal diffusion model (unlike ViNT) -- I would strongly recommend using this workflow over ViNT + image diffusion.

We do not have plans to release the heuristic stuff anytime soon due to multiple system-level dependencies, and this repo will likely only house the policy training and deployment code.

If you are interested in pointers, some of it has been reproduced and open sourced by another research group: https://github.com/UT-ADL/milrem_visual_offroad_navigation/

closed 2024-01-27 by robodhruv

chungcode0218, 2024-02-17:
Dear @robodhruv ,

I have a couple of questions regarding their navigation capabilities and would greatly appreciate your insights:

1. Navigation Without GPS/Satellite Assistance: Can ViNT and NoMaD navigate effectively without any GPS or satellite assistance? In such scenarios, is it correct to assume that the navigation process would be akin to exploring the environment from scratch until the target is located?

2. Reuse of Topological Maps in Explored Environments: If the model is reintroduced into an environment it has previously explored, would the existing topological maps still be applicable and useful, even if the starting point within the environment is different from before?

Thank you for your time and for sharing your valuable work with the community. I look forward to your response.