robodhruv/visualnav-transformer #16: Seeking Guidance for Simulating Models on Carla

chungcode0218, 2024-02-01:
Dear authors and respected experts,

I am currently looking to simulate models on Carla and I'm seeking guidance. Here's my current environment setup:

- Ubuntu 20.04
- Carla 0.9.13
- ROS Noetic

I believe I have successfully set up the Carla-Ros-Bridge and Kobuki. However, I'm unsure about how to integrate them effectively with Carla. Do I need to manually create cameras and robots (e.g., Turtlebot2) in Carla by myself, and create nodes to receive image data from the virtual cameras in the CARLA environment?

As I am relatively new to this field, I apologize in advance if my questions come across as rudimentary or offensive. Your assistance would be greatly appreciated.

Thank you.

Dashora7, 2024-02-02:
We used the following repo for CARLA experiments: https://github.com/OATML/oatomobile . This one is fully ROS-agnostic and build a gym-like API on top of CARLA. This system directly spawns sensors, pedestrians, cars, etc via API calls to the simulator's python API wrapper. If you use the oatomobile wrapper, you can mostly work abstracted above sensor management, but you can also add/remove things if needed.

closed 2024-02-02 by Dashora7

Dashora7, 2024-04-05:
Getting vehicle controls involves using CARLA's API for determining throttle and brake commands. This API runs through a PID controller which accepts a target speed and a desired waypoint. Use `from agents.navigation.controller import VehiclePIDController` to get the controller. One option is to simply convert your velocities into this format and plug it into the controller to generate an action usable for the agent.

942411526, 2024-04-05:
> > linear and angular
> 
> we have linear and angular without waypoints

> Getting vehicle controls involves using CARLA's API for determining throttle and brake commands. This API runs through a PID controller which accepts a target speed and a desired waypoint. Use `from [...]

convert to throttle and brake commands show a bad performer than others

Dashora7, 2024-04-05:
You can generate waypoints by assuming a time delta.

942411526, 2024-04-08:
> You can generate waypoints by assuming a time delta.

how to do this ? can you get me some ideas or guidance?

Dashora7, 2024-04-08:
Well using some small time delta, you can compute the angular displacement of the vehicle by multiplying angular velocity times time. In that direction you can then assume linear movement magnitude equal to time times the linear velocity. This should give a rough estimate of where it is. It's not perfect, you'd need derivatives, but a small enough time delta should fix that up.