robodhruv/visualnav-transformer #22: Waypoint Visualization and FOV Adjustment Effects using NoMaD Model in CARLA

chungcode0218, 2024-04-14:
Hello, 

I am currently working with the NoMaD model in the CARLA simulation environment and have encountered a couple of questions and issues that I hope you can help clarify.

**Waypoint Visualization :**
I have generated some visualization graphs for waypoints while using the NoMaD model. However, I am uncertain about the accuracy of these visualizations. Could you provide some guidance to confirm if my visualizations are correct?(It feels like the starting point is very far from the car)
[image]
[image]
[image]
Additionally, I have noticed that in some cases, the waypoints seem to scatter randomly. Is this a common issue? Will it affect the performance of the navigation?
[image]

**Field of View (FOV) Adjustment:**
In previous discussions, there was a mention that increasing the FOV might enhance navigation performance. I attempted to adjust the FOV from 90 to 120 to see its impact. After making this adjustment, the visualizations of the waypoints appeared somewhat unusual. Could this change in appearance be expected with a larger FOV? Additionally, could you confirm whether a larger FOV is indeed recommended for better navigation performance?
[image]
[image]

Any insights or suggestions you could provide would be greatly appreciated.

Thank you for your support.

laozheng1, 2024-05-31:
Excuse me, I'm new to this field. After following the readme tutorial step by step on ubantu, how do I migrate to the emulator?

laozheng1, 2024-06-12:
How did you achieve this visualization? During my run, I found that these waypoints were normalized coordinates and my visualization was poor.

chungcode0218, 2024-06-12:
Hi~Do you have WeChat?
Chinese is Ok?
我們可以討論一下

laozheng1, 2024-06-12:
> […](#)

可以的，非常感谢。
我的微信是：
szbl129

laozheng1, 2024-06-13:
> 我将该模型部署在服务器中，使用nomad进行路径预测，但是路径预测的效果不是很好，我可以加入讨论吗？
> 
> ![e2652276048aa5b6a5ba07ba4ec45d5](https://private-user-images.githubusercontent.com/72424705/338981200-b08fbddb-540c-4325-b5aa-2eff12e2b71a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyNDI3NTYsIm5iZiI6MTcxODI0MjQ1NiwicGF0aCI6Ii83MjQyNDcwNS8zMzg5ODEyMDAtYjA4ZmJkZGItNTQwYy00MzI1LWI1YWEtMmVmZjEyZTJiNzFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDAxMzQxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmMWE4MTMxMWIzMjVkODc0NmJjYzYxZmY1NzM2YzRiYWIwZmE1NDAwNzhlZTljMTU5YzVlZTg0OTE3ZmZiMjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VXkSFyo1vusS0fVUxby0h12aXGVklRdWEEi8Hq3oRfw)
> 
> > Hi~Do you have WeChat? Chinese is Ok? 我們可以討論一下 郑孙健 _**@**_._**>於 2024年6月12日 週三，下午5:13寫道：
> > […](#)

当然可以，加我加我。或者我加你也行

kjctphr, 2024-07-02:
Hi~ I am keen to know about how to generate the visualization graph, by any chance, can anyone share it to me? Greatly appreciate it.
你好，我也在路径显示上出了些问题，想向各位请教一下，方便加个微信吗？我是kjctphr。十分感激。

robodhruv, 2024-07-14:
@chungcode0218 
The visualization actually looks quite nice! I suspect there might be a small error in your camera matrix that is making the starting point offset from the bottom of the frame. You should be able to manually debug it to ensure `[0, 0]` maps to `[0, w/2]` after the transform.

The issue with the waypoint scattering likely points to the model performing poorly in some scenarios. Since the model is used via sampling, unless a majority of samples are so noisy, the closed-loop performance should be fine. Good luck testing!

robodhruv, 2024-07-14:
Closing this thread for now, but please feel free to reopen with questions!

closed 2024-07-14 by robodhruv

LebinLiang, 2024-08-20:
> Hi~ I am keen to know about how to generate the visualization graph, by any chance, can anyone share it to me? Greatly appreciate it. 你好，我也在路径显示上出了些问题，想向各位请教一下，方便加个微信吗？我是kjctphr。十分感激。
你好，我也在尝试可视化路径，可以加个微信讨论吗？

jimpap1, 2024-09-17:
Hi.I have trained the model but I don't own a robot so this must be done only with coding.
Can I feed an image from my dataset into the model and get an output like the images above with Waypoint Visualization? 
Thanks in advance!