robodhruv/visualnav-transformer #27: Error in train.py

jimpap1, 2024-05-10:
When i run the train.py i get errors.First i get torch and torchvision error.If i solve this i get number of workers error and then expected Tensor as element 0.What did i do wrong and can't train the dataset?

ajaysridhar0, 2024-07-14:
Hi @jimpap1,

Can you explain how you downloaded and processed the datasets? It's hard to pinpoint why the images failed to load without more context.

Thanks,
Ajay

swx153, 2024-07-17:
试试把.yaml文件里数据集的绝对路径换成相对路径
[image]

PEACHTTT, 2024-07-20:
> Hi @jimpap1, 你好
> 
> Can you explain how you downloaded and processed the datasets? It's hard to pinpoint why the images failed to load without more context.您能解释一下您是如何下载和处理数据集的吗？如果没有更多上下文，很难确定为什么图像无法加载。
> 
> Thanks, 谢谢 Ajay 阿杰

@ajaysridhar0 Hello,I met same problems about"Faild to load and TypeError".I download the datasets through the link of **README**.Then I processed it with **data_split.py**(I only tried the goStanford datasets),the change the directory in the vint.yaml

PEACHTTT, 2024-07-20:
> 试试把.yaml文件里数据集的绝对路径换成相对路径 [...]

尝试了一下没有用，请问你是没有遇到这个问题训练代码跑通了吗

swx153, 2024-07-22:
> 尝试了一下没有用，请问你是没有遇到这个问题训练代码跑通了吗

我之前遇到了其他问题，试着把这里的路径改了之后，就出现了无法加载图片的问题，然后我又改回去了。我已经跑通了

PEACHTTT, 2024-07-25:
@swx153 感谢回复！我还想确认一下关于stanford的那个数据集，是不是只需要data_split.py划分处理。我有搜到pytorch版本问题也会导致这个原因，可以问一下你的torch版本吗？方便的话能不能加一下联系方式，我想对比下是哪一步出了问题。目前我卡在这一步
[image]

swx153, 2024-07-26:
@PEACHTTT stanford这个数据集确实只需要划分，我的torch是2.3.1，我的微信号是swxdaydayup

swx153, 2024-07-30:
@PEACHTTT standford数据集确实只需要split操作，我的torch版本是2.3.1，如果有需要可以加我微信swxdaydayup

PEACHTTT, 2024-07-30:
> @PEACHTTT standford数据集确实只需要split操作，我的torch版本是2.3.1，如果有需要可以加我微信swxdaydayup
十分感谢！抱歉我暂时不参与这个相关项目了，谢谢你的耐心回复！

PEACHTTT, 2024-07-30:
@swx153 十分感谢！抱歉我暂时不参与这个相关项目了，谢谢你的耐心回复！

jimpap1, 2024-08-02:
Why do i get Aborted (core dumped) when i run train.py ?

plusgrey, 2024-08-07:
> Why do i get Aborted (core dumped) when i run train.py ?

Same with torch 2.4.0, ubuntu 18.04

plusgrey, 2024-08-08:
> > Why do i get Aborted (core dumped) when i run train.py ?
> 
> Same with torch 2.4.0, ubuntu 18.04

I have solved the problem by downgrading the opencv to 4.1.2.30. I think two main reasons will cause the "Aborted (core dumped)" error. 1: Too small image_log_freq (>0) or too large num_image_log, these settings takes lots of memory which might cause this error. 2: In some machine, the version of PyQT5 mismatches the opencv-python version used in ViNT, this will also cause the error.   p.s. sometimes, the torch.multiprocessing can also cause this error.

LeZheng-x, 2024-12-12:
Maybe you should delete dataset_xx.lmdb file, and try run the train.py again. I found the code once the detection cache file is created, it will not be updated.You can try to use gdb to output whether _image_cache has reasonable key-value pairs in vint_dataset.py. Hope it will help you!