robodhruv/visualnav-transformer #29: Questions about the training

xzxzmmnn, 2024-05-29:
Thank you for providing the code. I have successfully trained the model from scratch using the public dataset you shared. The training process seemed to be successful; I've monitored the loss decrease and the visualization results in wandb appear to be satisfactory.

However, when I run the explore.py code using the model checkpoint file from my training, the model outputs are incorrect. To verify the functionality of explore.py, I also executed the code using the model checkpoint file you provided, which worked perfectly fine.

So, I am wondering if I need to modify the explore.py code to better utilize the model that I trained from scratch?

liumingzs, 2024-06-14:
I want to know how many datasets you download,Can I only download the recon to train the model?

ajaysridhar0, 2024-07-03:
@xzxzmmnn Can you describe how the model outputs are incorrect?
@liumingzs You should be able to download all the datasets [here](https://github.com/robodhruv/visualnav-transformer?tab=readme-ov-file#data-wrangling). Let me know if you run into any specific issues.

spokV, 2024-07-10:
It seems that TartanDrive is down. Nothing at http://airlab-share.andrew.cmu.edu/dataset-icra22/..

robodhruv, 2024-07-14:
I recommend creating an issue at the relevant repo: https://github.com/castacks/tartan_drive or to the authors for assistance.