robodhruv/visualnav-transformer #30: Failed to load image

jimpap1, 2024-06-02:
Why does it fail to load images from RECON dataset? 
[image]

laozheng1, 2024-06-03:
Are you in the training phase? Does the code in the later deployment phase run? I've been running this code recently, and I've had some problems.

jimpap1, 2024-06-03:
Yes it is in the training part.I haven't moved yet to the deployment part as i'm trying to solve the training errors,

robodhruv, 2024-07-14:
Did you download the data in the appropriate folders first? Maybe there is a small bug in the image paths?

ajaysridhar0, 2024-07-14:
Let's move our discussion to here: https://github.com/robodhruv/visualnav-transformer/issues/27.

closed 2024-07-14 by ajaysridhar0