robodhruv/visualnav-transformer #35: Go1 deployment details

hchlhwang, 2024-07-11:
Thank you for the great work!

Would it be possible to share scripts relevant to deploying the code on the Go1 robot (e.g., robot.yaml, topomap create code)? 

Thank you.

ajaysridhar0, 2024-07-14:
Hi @hchlhwang,

I think this might be useful: https://github.com/robodhruv/visualnav-transformer/issues/6#issuecomment-1911494770. Also, check out the section for scripts to generate a topomap: https://github.com/robodhruv/visualnav-transformer/issues/6#issuecomment-1911494770.

--Ajay

closed 2024-07-14 by ajaysridhar0

hchlhwang, 2024-09-23:
Can you share the go1.yaml params that worked best for go1? Also, can you reshare the link to creating the topomap? I think your link refers to the go1-run script. Thanks!