robodhruv/visualnav-transformer #36: Error in train.py

zqs010908, 2024-07-13:
Thank you for providing the code.
I am trying to use train.by to train my model，but I encountered the following issue while using train.py

Traceback (most recent call last):
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/train.py", line 402, in <module>
    main(config)
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/train.py", line 326, in main
    train_eval_loop_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/vint_train/training/train_eval_loop.py", line 196, in train_eval_loop_nomad
    ema_model = EMAModel(model=model,power=0.75)
TypeError: __init__() missing 1 required positional argument: 'parameters'

Traceback (most recent call last):                                                                                                                                                                                                                                              
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/train.py", line 402, in <module>
    main(config)
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/train.py", line 326, in main
    train_eval_loop_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/vint_train/training/train_eval_loop.py", line 203, in train_eval_loop_nomad
    train_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/vint_train/training/train_utils.py", line 661, in train_nomad
    loss.backward()
  File "/home/iiau-vln/miniconda3/envs/nomad/lib/python3.8/site-packages/torch/_tensor.py", line 522, in backward
    torch.autograd.backward(
  File "/home/iiau-vln/miniconda3/envs/nomad/lib/python3.8/site-packages/torch/autograd/__init__.py", line 266, in backward
    Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
RuntimeError: element 0 of tensors does not require grad and does not have a grad_fn

zolmaeng, 2024-07-13:
Try using `EMAModel(model.parameters(), power=0.75)` instead of `EMAModel(model, power=0.75)`.

robodhruv, 2024-07-14:
Thanks @zolmaeng!  Does that fix the problem, @zqs010908 ?
We also welcome PRs with reproducible bug reports :)

zqs010908, 2024-07-14:
I have already used EMAModel(model.parameters(), power=0.75)，but a new problem has arisen

/home/iiau-vln/miniconda3/envs/nomad/lib/python3.8/site-packages/diffusers/training_utils.py:361: FutureWarning: Passing a `torch.nn.Module` to `ExponentialMovingAverage.step` is deprecated. Please pass the parameters of the module instead.
  deprecate(
Traceback (most recent call last):                                                                                                                                                                                                                                                                         
  File "/home/iiau-vln/ws_zqs/nomad/ori_nomad/visualnav-transformer/train/train.py", line 402, in <module>
    main(config)
  File "/home/iiau-vln/ws_zqs/nomad/ori_nomad/visualnav-transformer/train/train.py", line 326, in main
    train_eval_loop_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/ori_nomad/visualnav-transformer/train/vint_train/training/train_eval_loop.py", line 203, in train_eval_loop_nomad
    train_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/ori_nomad/visualnav-transformer/train/vint_train/training/train_utils.py", line 676, in train_nomad
    ema_model.averaged_model,
AttributeError: 'EMAModel' object has no attribute 'averaged_model'

I found that the EMAModel class in diffusers.training_utils indeed does not have the averaged_model object. I see that the EMAModel in diffusion_policy has the averaged_model object, as referenced in line 31 of ema_model.py. I'm not sure if this method is correct.

zqs010908, 2024-07-15:
And after I used the EMAModel in diffusion_policy, I found that it solved the previous problem, but now a new issue has arisen. I am using the config from nomad and training with the SACSoN/HuRoN dataset. Have you encountered this issue before?
Traceback (most recent call last):                                                                                                                                                                                                                                                             
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/train.py", line 402, in <module>
    main(config)
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/train.py", line 326, in main
    train_eval_loop_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/vint_train/training/train_eval_loop.py", line 203, in train_eval_loop_nomad
    train_nomad(
  File "/home/iiau-vln/ws_zqs/nomad/visualnav-transformer/train/vint_train/training/train_utils.py", line 860, in train_nomad
    loss.backward()
  File "/home/iiau-vln/miniconda3/envs/nomad/lib/python3.8/site-packages/torch/_tensor.py", line 522, in backward
    torch.autograd.backward(
  File "/home/iiau-vln/miniconda3/envs/nomad/lib/python3.8/site-packages/torch/autograd/__init__.py", line 266, in backward
    Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
RuntimeError: element 0 of tensors does not require grad and does not have a grad_fn

keshav0306, 2024-07-26:
Change line 31 of the file ema_model.py in diffusion_policy/diffusion_policy/model/diffusion/ from
 `self.averaged_model = model` to  `self.averaged_model = copy.deepcopy(model)`

SkyDream666, 2025-03-12:
change the diffusers version to: pip install diffusers==0.11.1

czhuhauhdguga, 2025-03-12:
I have the same problem with "ema_model.averaged_model",but when i did:pip install diffusers==0.11.1,there is a error:
(vint_train) wyy@flashwyy:~/visualnav-transformer-main/train$ python [train.py](http://train.py/) -c ./config/nomad.yaml
Traceback (most recent call last):
File "[train.py](http://train.py/)", line 17, in <module>
from diffusers.schedulers.scheduling_ddpm import DDPMScheduler
File "/home/wyy/anaconda3/envs/vint_train/lib/python3.8/site-packages/diffusers/init.py", line 46, in <module>
from .pipeline_utils import DiffusionPipeline
File "/home/wyy/anaconda3/envs/vint_train/lib/python3.8/site-packages/diffusers/pipeline_utils.py", line 35, in <module>
from .dynamic_modules_utils import get_class_from_dynamic_module
File "/home/wyy/anaconda3/envs/vint_train/lib/python3.8/site-packages/diffusers/dynamic_modules_utils.py", line 29, in <module>
from huggingface_hub import HfFolder, cached_download, hf_hub_download, model_info
ImportError: cannot import name 'cached_download' from 'huggingface_hub' (/home/wyy/anaconda3/envs/vint_train/lib/python3.8/site-packages/huggingface_hub/init.py)

czhuhauhdguga, 2025-03-13:
Thank you,it works well

changshu2002, 2025-06-10:
That's very good, and it has well solved my problem.