robodhruv/visualnav-transformer #39: NoMad on Carla

reachpranjal, 2024-08-08:
Hi @robodhruv and @ajaysridhar0 

I'm trying to deploy NoMad on Carla under closed-loop setting with provided checkpoint fine-tuned in the Carla distribution. However, it doesn't seem to be working well even for a straight run (no traffic). The normalized waypoints plot doesn't seem to be apt. 

Providing the script. 
https://gist.github.com/reachpranjal/ca77e69c338dcbd4388cb441f5eb8f8d

ajaysridhar0, 2024-08-13:
I can't access the script. It might be because it's part of a private repo. Can you share it another way?

closed 2024-08-14 by reachpranjal

plusgrey, 2025-01-16:
> Thanks for your reply. I managed to fix it. Data is the key :)

Hi, could you please share your script as I also have problems when deploying ViNT and NoMad on CARLA. Thanks so much/

plusgrey, 2025-01-16:
> Thanks for your reply. I managed to fix it. Data is the key :)

Hi @reachpranjal , could you please share your script as I also have problems when deploying ViNT and NoMad on CARLA. Thanks so much.

zljjacob, 2025-05-26:
@reachpranjal Could you explain the key of data? I don't know what factors should I consider when collecting data? Thank you.