robodhruv/visualnav-transformer #43: training stopped because of Qt?

hszhoushen, 2024-09-13:
When training diffusion policy on the remote server, it encountered one problem as follows, which indicates the training is aborted, because of the move to the target thread and Qt platform. How do we solve this issue?

Using cosine annealing with T_max 100
Using warmup scheduler
Start ViNT DP Training Epoch 0/99
Train Batch:   0%|                                                | 0/1434 [01:06<?, ?it/s, loss=1.04](epoch 0) (batch 0/1433) uc_action_loss (train): 42.5554 (100pt moving_avg: 42.5554) (avg: 42.5554)
(epoch 0) (batch 0/1433) uc_action_waypts_cos_sim (train): 0.4572 (100pt moving_avg: 0.4572) (avg: 0.4572)
(epoch 0) (batch 0/1433) uc_multi_action_waypts_cos_sim (train): 0.5289 (100pt moving_avg: 0.5289) (avg: 0.5289)
(epoch 0) (batch 0/1433) gc_dist_loss (train): 150.1345 (100pt moving_avg: 150.1345) (avg: 150.1345)
(epoch 0) (batch 0/1433) gc_action_loss (train): 44.8862 (100pt moving_avg: 44.8862) (avg: 44.8862)
(epoch 0) (batch 0/1433) gc_action_waypts_cos_sim (train): 0.4524 (100pt moving_avg: 0.4524) (avg: 0.4524)
(epoch 0) (batch 0/1433) gc_multi_action_waypts_cos_sim (train): 0.5219 (100pt moving_avg: 0.5219) (avg: 0.5219)
QObject::moveToThread: Current thread (0x44de4a70) is not the object's thread (0x450a0d00).
Cannot move to target thread (0x44de4a70)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/xxx/nomad_train/lib/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, minimal, minimalegl, offscreen, vnc, webgl.

Aborted (core dumped)

hszhoushen, 2024-09-14:
pip uninstall opencv-python
pip install opencv-python-headless

it worked for me
https://github.com/NVlabs/instant-ngp/discussions/300

tangyipeng100, 2024-11-05:
> pip uninstall opencv-python pip install opencv-python-headless
> 
> it worked for me [NVlabs/instant-ngp#300](https://github.com/NVlabs/instant-ngp/discussions/300)

it works, thanks.