robodhruv/visualnav-transformer #44: Waypoint Visualization (without robot deployment)

jimpap1, 2024-09-17:
Hi and sorry for making an issue.
I have trained the model but I don't own a robot so this must be done only with coding.
Can I feed an image from my dataset into the model and get an output on my image with Waypoint Visualization?
Thanks in advance!

kjctphr, 2024-09-18:
yes you can, just replace the input with opencv instead.

jimpap1, 2024-09-18:
> yes you can, just replace the input with opencv instead.

How could i do that? In which file should i replace it? Any more help? Thanks!

jimpap1, 2024-09-19:
> deployment/navigate.py, Chatgpt is your best companion

I managed to get waypoints from navigate.py. Now how do i visualise them on my images?

Benaidandan, 2024-09-24:
> > deployment/navigate.py, Chatgpt is your best companion
> 
> I managed to get waypoints from navigate.py. Now how do i visualise them on my images?

Did you do that? Does that have to be deployed on the real robot？

jimpap1, 2024-09-24:
> > > deployment/navigate.py, Chatgpt is your best companion
> > 
> > 
> > I managed to get waypoints from navigate.py. Now how do i visualise them on my images?
> 
> Did you do that? Does that have to be deployed on the real robot？

Yes I got some waypoints from an image dataset without robot.Now i want to visualise these trajectories on the images from the dataset.Using function plot_trajs_on_image needs some arguments that arent computed.

caokaifa, 2025-02-17:
@jimpap1，Have you solved this problem? I also want to know how this problem was resolved.