robodhruv/visualnav-transformer #52: Is it possible to deploy without hardware?

binichallein, 2024-11-26:
My environment is WSL with Ubuntu 20.04. After installing ROS Noetic, I realized that Kobuki requires a real physical device to connect to. However, I don’t have access to the actual hardware. Is it possible to use simulation software instead? Do you have any recommendations? I’m a beginner with ROS and not very familiar with the hardware aspects. I hope this doesn’t trouble you. Thank you!

Mrakas, 2024-12-13:
try this https://github.com/Mrakas/nomad/blob/main/deployment/src/script/api_test_RGB.py

forresteh, 2025-02-24:
same question

Qhijane, 2025-06-30:
> try this https://github.com/Mrakas/nomad/blob/main/deployment/src/script/api_test_RGB.py

Recommended Navigation Waypoint: [ 0.375 -0.2  ]    Only output waypoints without outputting the visualization results?