robodhruv/visualnav-transformer #64: Action stats

zxp771, 2025-04-08:
# global params for diffusion model
# normalized min and max
action_stats:
  min: [-2.5, -4] # [min_dx, min_dy]
  max: [5, 4] # [max_dx, max_dy]

Thank you for sharing. There is a question about action stats parameter design, why such a set of great and small values of dx and dy are set. What is the rationale behind it? Or is it dependent on the motion characteristics of the platform at the time of data collection. In other words, if you apply your work to a different platform, do you need to adjust this set of parameters.

hjr37, 2025-04-28:
Hi, I have the same problem with this parameter group, have you solved it?

zxp771, 2025-05-13:
> Hi, I have the same problem with this parameter group, have you solved it?

It's still a problem for me. 
With our understanding of the code, here should be the added constraint that the label loaded from the dataset contains information about the action. So far, it should be constraining the predicate action to a variable amount in the x and y direction. My understanding is that if the actual application platform speed exceeds this constraint amount, then this parameter value should be invalidated. Of course, the ground and robots in general don't seem to be able to generate such large displacements between each waypoint (given that some of the parameters in /data/data_config)) are set in meters.

[image]

duyifan02, 2025-06-26:
I’m working on porting this work to different platforms. Their approach has issues: the waypoints don’t convey distance information well, and when you reduce the step‑forward distance to a very low value, the obstacle‑avoidance performance improves dramatically. In other words, in some cases it closely resembles behavior cloning.